Ericsii/FAST_LIO_ROS2 #5: Time synchronization problem

wyb17js, 2024-01-09:
Hello author, I have experimented with your project on humble. My laser device is mid360, which has its own imu, but I don't know how to align the time stamps of the two, which causes my positioning to be easily lost and the odometry to drift.I noticed that you mentioned in the readme that in-program synchronization is not guaranteed to be valid. So I would like to ask if you have any good suggestions?

wyb17js, 2024-01-09:
Sorry, I found that I seemed to ask you similar questions about dynamic time migration before, but you didn't have any good opinions, so I would like to know how I can judge whether my radar and imu timestamps are aligned through rosbag? Shouldn't there be multiple imu messages between the two frames?

Ericsii, 2024-01-16:
Look no new updates. Close the issue

closed 2024-01-16 by Ericsii