Ericsii/FAST_LIO_ROS2 #9: Point Cloud Output "Fuzzy" and Path is not correct

ryanpdoc57, 2024-01-18:
I am using ROS2 Foxy to run the package. Pictured below is the Velodyne bag file output in RViz for  "20120615-001", however, the same result happens no matter what LiDAR type/bag file I use. The output is extremely "fuzzy" and shows a significantly lower-quality image as the ROS1 version of the same bag file. Also, the path is not completely accurate, as seen in the images below. There is some drift in Odometry that appears in the ROS2 version, but not the ROS1 version. Does anyone have any input on this? The first Image is ROS1, while the second image is ROS2. Thanks in advance!
[image]
[image]

Ericsii, 2024-01-19:
I did not migrate the RVIZ config from the original version. The RVIZ config is a handmade one. Due to FASTLIO being only a lidar odometry if the trajectory is correct then everything is working correctly.

You should disable the pointcloud view in RVIZ and compare only the trajectories generated from ROS1 and ROS2.

ryanpdoc57, 2024-01-19:
As you can see in the following picture, the path is also not correct. Measuring with a protractor, the angle of the path using ROS2 (left image) is about 30 degrees, while the angle for ROS1 (right image) is about 40 degrees. The red line being flat in both images (a wall or something like that) proves the difference in the angles.  The trajectories should be the exact same. Have you done any data set testing that proves the full functionality of ROS2 FAST-LIO? Through my testing using different data sets, I have reason to believe the package does not function as expected. I would love to be wrong though. 
[image]

Ericsii, 2024-01-20:
Could you please test it on some open-source datasets with ground-truth odometry? In this way, we can compare the effects of these two versions.

ryanpdoc57, 2024-01-20:
> Could you please test it on some open-source datasets with ground-truth odometry? In this way, we can compare the effects of these two versions.

I am trying to use the M2GDR data set (https://github.com/SJTU-ViSYS/M2DGR?tab=readme-ov-file) which contains ground truth information. I am using plot juggler to plot the trajectory of the odometry from FAST-LIO, but I don't know how to overlay this plot with the ground truth (.txt file). Do you have any advice on how to do this? The bag files are too big for Matlab.

Ericsii, 2024-01-22:
You can use [EVO](https://github.com/MichaelGrupp/evo). This is a well-used toolbox for SLAM evaluation

Ericsii, 2024-02-20:
Close for no update

closed 2024-02-20 by Ericsii

ginwodka, 2024-09-05:
I can confirm this.
Something is different compared to original FAST LIO 2 version with ROS1. First of all, you have a spare pointcloud. It doesnt matter if i have point filter num = 1. Thats definitely not the case in ROS1. Second, i have difficulties while mapping indoor and outdoor. Most of the time the SLAM fails. And i mapped a lot of objects with Fast Lio 2 and ROS1 and there were no problems at all. So i assume this version here is not working properly :(