Ericsii/FAST_LIO_ROS2 #10: Is ouster lidar available?

LTHlidar, 2024-01-25:
First, thank you for your FAST LIO ROS2 package.

I have a question, I am currently using ouster lidar.
To launch a package using ouster lidar, 
` ros2 launch fast_lio mapping_ouster64.launch` command into the terminal? 
Or, inside the mapping.launch.py file, modify the .yaml part to ouster64.yaml and run the command
`ros2 launch fast_lio mapping.launch.py` and run it?

Actually, I tried both ways, but it didn't work, hence the question.

Thanks.

Ericsii, 2024-01-28:
Other types of lidar support will be updated in recent days

LTHlidar, 2024-01-29:
I successfully ran fast LIO2 using the Outster Rider topic. 

in laserMapping.cpp 
`sub_pcl_pc_ = this->create_subscription<sensor_msgs::msg::PointCloud2>(lid_topic, 10, standard_pcl_cbk)` 
need to be fixed to
`sub_pcl_pc_ = this->create_subscription<sensor_msgs::msg::PointCloud2>(lid_topic, rclcpp::SensorDataQoS(), standard_pcl_cbk)`

If the QOS settings of lidar publish topic and fast lio subscriber are different(ex : RELIABILITY or Best_effort), the outer topic is not subscribed.
I haven't analyzed the data exactly, but it seems to work well with ouster lidar.

LTHlidar, 2024-01-29:
And I don't know why, but I need to paste the code from mid360.yaml file into ouster64.yaml and replace it with the right code for ouster64.yaml, then fast lio 2 will run without error.
I'm not an expert, so I don't know the exact solution, but I'm commenting in case it helps. Thanks (I think it's probably due to the different systems the files were created on).

Ericsii, 2024-01-31:
> And I don't know why, but I need to paste the code from mid360.yaml file into ouster64.yaml and replace it with the right code for ouster64.yaml, then fast lio 2 will run without error.

This is because I only migrated mid360.yaml to ros2, ouster and other lidars are not migrated yet.

Ericsii, 2024-02-20:
Other lidar support is added. You can have a try.

closed 2024-02-20 by Ericsii