Ericsii/FAST_LIO_ROS2 #15: Save map takes too long or freezed

TZECHIN6, 2024-07-08:
[image]

First of all, this rosbag is recorded from the gazebo and the simulated LiDAR is the MID360 from Livox. When I triggered the service, it said saving map... however, I have waited for a while and it still at saving stage. If I close the FAST-LIO, the map will be lose.

I believe the data size of the rosbag is not very big, just few minutes. And when I use the real LiDAR, the map can be saved without issue.

I have closed the simulator and just let the rosbag to play.

Ericsii, 2024-07-22:
Could you provide your bag file? So that I can test it on my environment

TZECHIN6, 2024-07-23:
Let me find the bag and send it to you. Any way I could share it to you?

Ericsii, 2024-07-24:
You can use any cloud storage you like to share

TZECHIN6, 2024-08-07:
I have reopened this issue. The save map takes very long (I believes it is freezed)

```
thomas@UN1290-059:~/Code/FAST-LIO_ws$ ros2 launch fast_lio mapping.launch.py config_file:=mid360.yaml
[INFO] [launch]: All log files can be found below /home/thomas/.ros/log/2024-08-07-10-20-58-199187-UN1290-059-48972
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fastlio_mapping-1]: process started with pid [48973]
[INFO] [rviz2-2]: process started with pid [48975]
[rviz2-2] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[fastlio_mapping-1] [INFO] [1722997258.329762274] [laser_mapping]: p_pre->lidar_type 1
[fastlio_mapping-1] Multi thread started 
[fastlio_mapping-1] ~~~~/home/thomas/Code/FAST-LIO_ws/src/FAST_LIO_ROS2/ file opened
[fastlio_mapping-1] [INFO] [1722997258.335711391] [laser_mapping]: Node init finished.
[fastlio_mapping-1] IMU Initial Done
[fastlio_mapping-1] [WARN] [1722997258.485259771] [laser_mapping]: No point, skip this scan!
[fastlio_mapping-1] 
[rviz2-2] [INFO] [1722997258.508096766] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1722997258.508150899] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1722997258.523257739] [rviz2]: Stereo is NOT SUPPORTED
[fastlio_mapping-1] [INFO] [1722997258.586109962] [laser_mapping]: Initialize the map kdtree
[fastlio_mapping-1] [INFO] [1722997271.342311890] [laser_mapping]: Saving map to ./test.pcd...
```

@Ericsii The ros2bag is uploaded to below link, please let me know if you could not download it. Great thanks for your help.
https://1drv.ms/f/s!AvsFv5difaf3iSF8UsgJ3Xz07Thh?e=TpGmAT

TZECHIN6, 2024-08-12:
Sorry for keep pushing this post... I am not sure which part of the setup is wrong that lead to such very long time or freezed saving (cannot be saved). I have tried to install the official FAST-LIO (ROS2) and the livox_ros_driver2, with minimal config changed (ip address and multi-topic). Still cannot save the map successfully.

Ericsii, 2024-08-12:
You can try to add a voxel grid downsample filter before saving the map in this function:

https://github.com/Ericsii/FAST_LIO_ROS2/blob/b6ebca01663793ad5faa55324d7af7576ed8bb2e/src/laserMapping.cpp#L609

BTW the official FAST-LIO ROS2 version is merged from this repo, so the behavior of both should be the same.

TZECHIN6, 2024-08-12:
Thanks for the help... actually I found the .pcd file is generated, even through the console is saying `saving the map...`

Is it normal or the message should be `map save done`?

Ericsii, 2024-08-15:
This is because the process of saving the pcd file is not complete yet

TZECHIN6, 2024-08-22:
But the result in pcl_viewer seem fine.... well, I will close this issue for now. If the generated SLAM pointcloud could not be used, I will reopen this later on. Thanks.

closed 2024-08-22 by TZECHIN6