Ericsii/FAST_LIO_ROS2 #18: Is it able to use PointCloud2 for Livox mid360?

TZECHIN6, 2024-08-03:
I have created a custom PointCloud2 struct in my livox ros driver2, the reason for doing so is to utilise other point-cloud preprocessing node features, like crop box filter and concatenation. (Cropbox is essential here, as currently the vehicle body is reflecting and creating noise when mapping)

I found the `mid360_handler` function takes PointCloud2 as input rather than CustomMsg. And there are also Ouster and velodyne which do not publish the offset time from base time in their driver.

The question here is, for my purpose of cleaning up those static noises, Should I implement the crop box feature using CustomMsg, or any quick and simple way say increase the minimal distance (I did not find this setting btw) so that it ignores the vehicle body points?

Ericsii, 2024-08-06:
In [livox_ros_driver2](https://github.com/Livox-SDK/livox_ros_driver2), there is also a launch parameter `xfer_format` that can easily switch the output pointcloud format from CustomMsg to PointCloud2.

For the general PointCloud2 input format, you can change the `lidar_type` to `4`, using the `Preprocess::default_handler`, but without pointcloud undistortion.
https://github.com/Ericsii/FAST_LIO_ROS2/blob/b6ebca01663793ad5faa55324d7af7576ed8bb2e/config/mid360.yaml#L20

The `Preprocess::mid360_handler` is currently deprecated due to the reusable of the `Preprocess::avia_handler`

TZECHIN6, 2024-08-06:
The reason why I modified the PointCloud2 struct is that I am integrating livox mid360 to an autonomous robot project which utilize autoware, and it gives a lidar preprocessing workflow (details can be found in [here](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/sensing/data-types/point-cloud/#overview)). The self-cropbox is very useful for my use case.

I would like to utilize the pointcloud data from there, and integrate FAST-LIO mapping into it.

Ericsii, 2024-08-06:
> I would like to utilize the pointcloud data from there, and integrate FAST-LIO mapping into it.

In this case, you can change the configuration in `mid360.yaml`. Set `lidar_type` to 4 then it can process PointCloud2 data.

TZECHIN6, 2024-08-06:
Cool @Ericsii I will give it a try now. have a nice day :D

closed 2024-08-06 by Ericsii