Ericsii/FAST_LIO_ROS2 #19: No pointcloud output in Rviz running Livox Mid-360 on ROS 2 Humble

NicholasTanYY, 2024-08-29:
Hello @Ericsii , 

I'm trying to get the Livox Mid360 Lidar to run on my computer but am currently unable to get any output in Rviz. When I run the commands  `ros2 launch livox_ros_driver2 msg_MID360_launch.py` in the right terminal and `ros2 launch fast_lio mapping.launch.py config_file:=mid360.yaml` in the left terminal, I get the result shown in the following image, with nothing appearing in the Rviz display.

[image]

I have provided more details in [this](https://github.com/hku-mars/FAST_LIO/issues/354) issue in the main FAST_LIO repo. Would really appreciate any assistance!

Ericsii, 2024-09-03:
You are using a wrong configuration. Please replace the `mid360.yaml` by this new configuration file https://github.com/Ericsii/FAST_LIO_ROS2/blob/ros2/config/mid360.yaml

Ericsii, 2024-09-03:
> You are using a wrong configuration. Please replace the `mid360.yaml` by this new configuration file https://github.com/Ericsii/FAST_LIO_ROS2/blob/ros2/config/mid360.yaml

This is because you used a wrong `lidar_type`. Just simply set the `lidar_type` to 1 will also solve your problem

closed 2024-10-23 by Ericsii

maxxbolotov, 2024-11-27:
> > You are using a wrong configuration. Please replace the `mid360.yaml` by this new configuration file https://github.com/Ericsii/FAST_LIO_ROS2/blob/ros2/config/mid360.yaml
> 
> This is because you used a wrong `lidar_type`. Just simply set the `lidar_type` to 1 will also solve your problem

I have the same issue using Livox Mid 360. My config file is correct but there is only black screen in RVIZ.
Log and screenshot below:
```
~/src$ ros2 launch fast_lio mapping.launch.py config_file:=mid360.yaml
[INFO] [launch]: All log files can be found below /home/sageezz/.ros/log/2024-11-27-22-12-07-592806-ubuntu-74286
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fastlio_mapping-1]: process started with pid [74287]
[INFO] [rviz2-2]: process started with pid [74289]
[fastlio_mapping-1] [INFO] [1732734727.970661177] [laser_mapping]: p_pre->lidar_type 1
[fastlio_mapping-1] Multi thread started 
[fastlio_mapping-1] ~~~~/home/sageezz/src/ros_ws/FAST_LIO/ file opened
[fastlio_mapping-1] [INFO] [1732734727.992144964] [laser_mapping]: Node init finished.
[rviz2-2] [INFO] [1732734728.743846298] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1732734728.744100098] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1732734728.797618919] [rviz2]: Stereo is NOT SUPPORTED
```

[image]