Ericsii/FAST_LIO_ROS2 #22: 如何搭配Livox Avia来使用

Rorschach7771, 2024-10-17:
我在配好Livox AVIA对应的环境后，运行了livox_ros2_driver并且获得了AVIA的点云。但是该点云的消息格式为Type:livox_interfaces/msg/CustomMsg  。这与本项目中的消息格式不符，导致程序无法获得点云。我注意到本项目中处理原始点云的代码是        if (p_pre->lidar_type == AVIA)
        {
            sub_pcl_livox_ = this->create_subscription<livox_ros_driver2::msg::CustomMsg>(lid_topic, 20, livox_pcl_cbk);
        }。 请问如何才能做适配使得本代码可以搭配AVIA来运行呢？感谢！

Ericsii, 2024-10-23:
livox_interfaces/msg/CustomMsg 的格式应该是和livox_ros_driver2/msg/CustomMsg 一样的。你可以查看官方这两个repo的msg定义。因此直接将CMakeLists和package.xml里面的依赖改成livox_interfaces，同时代码里面include livox_interfaces/msg/CustomMsg 并将 你提到的这一行改成livox_interfaces/msg/CustomMsg类型即可

https://github.com/Livox-SDK/livox_ros2_driver/blob/249be8911010e201f620884ea9d8e7d133a37c0a/livox_interfaces/msg/CustomMsg.msg#L1-L8

https://github.com/Livox-SDK/livox_ros_driver2/blob/6b9356cadf77084619ba406e6a0eb41163b08039/msg/CustomMsg.msg#L1-L8

Rorschach7771, 2024-10-24:
> livox_interfaces/msg/CustomMsg 的格式应该是和livox_ros_driver2/msg/CustomMsg 一样的。你可以查看官方这两个repo的msg定义。因此直接将CMakeLists和package.xml里面的依赖改成livox_interfaces，同时代码里面include livox_interfaces/msg/CustomMsg 并将 [...]

非常感谢！

closed 2024-10-24 by Rorschach7771