Ericsii/FAST_LIO_ROS2 #23: ROS2 Foxy Compile failed : exited with code 2

tenmoonsninesky, 2024-11-05:
Starting >>> livox_ros_driver2
Starting >>> livox_sdk2
Finished <<< livox_sdk2 [0.17s]                                           
Finished <<< livox_ros_driver2 [0.50s]                     
Starting >>> fast_lio
Failed   <<< fast_lio [25.1s, exited with code 2]  

Summary: 2 packages finished [25.9s]
  1 package failed: fast_lio

[](https://github.com/hku-mars/FAST_LIO/pull/334)
After submitting this pull request, it still cannot be compiled. The error message does not provide a detailed explanation of where the error occurred.

tenmoonsninesky, 2024-11-06:
[  8%] Built target fast_lio__cpp
[ 11%] Built target fast_lio__rosidl_generator_c
[ 28%] Built target fast_lio__rosidl_typesupport_c
[ 31%] Built target fast_lio__rosidl_typesupport_cpp
[ 37%] Built target fast_lio__rosidl_typesupport_introspection_c
[ 45%] Built target fast_lio__rosidl_typesupport_introspection_cpp
[ 54%] Built target fast_lio__rosidl_typesupport_fastrtps_cpp
[ 62%] Built target fast_lio__rosidl_typesupport_fastrtps_c
[ 62%] Built target fast_lio
[ 65%] Built target fast_lio__py
[ 71%] Built target fast_lio__python
[ 74%] [32mBuilding CXX object CMakeFiles/fastlio_mapping.dir/src/laserMapping.cpp.o[0m
[ 91%] Built target fast_lio__rosidl_typesupport_c__pyext
[ 91%] Built target fast_lio__rosidl_typesupport_fastrtps_c__pyext
[ 91%] Built target fast_lio__rosidl_typesupport_introspection_c__pyext

Ericsii, 2024-11-06:
Please clean your workspace `log` `install` `build` and rebuild fast_lio. Then paste the log after your first build. In this way the log information should be complete

tenmoonsninesky, 2024-11-06:
I found the error, it was caused by manually modifying the code with Chinese spaces attached.

closed 2024-11-06 by tenmoonsninesky