Ericsii/FAST_LIO_ROS2 #24: Ouster OS1-128 Lidar : Frame [carema_init] does not exist.

tenmoonsninesky, 2024-11-06:
您好，感谢您之前的帮助，我已编译成功。

我的配置是
Ubuntu 20.04
ROS2 Foxy
Ouster OS1-128 128线激光雷达

现遇到以下问题：

在输入`ros2 launch fast_lio mapping.launch.py config_file:=ouster128.yaml` 后，会出现Global status warn，

[image]

但是在输入 `ros2 launch ouster_ros sensor.launch.xml  sensor_hostname:=169.254.16.131  viz:=false point_cloud_frame:=os_sensor  lidar_mode:=1024x10 point_type:=original` 开启Ouster-ros 驱动之后，

[image]

出现 Global status : error Frame [carema_init] does not exist.

Fast-lio的terminal如此显示

[image]

同时没有点云出现。

[image]

tf tree如此显示。

单独开启ouster-ros的rviz是能够显示点云的。但是fast lio似乎无法运行。

我自己修改的ouster128yaml如下：将topic的名字改为新版ouster-ros驱动的名字，scan_line改为128。fov_degree det_range extrinsic_T extrinsic_R 也进行了修改。

/**:
    ros__parameters:
        feature_extract_enable: false
        point_filter_num: 3
        max_iteration: 3
        filter_size_surf: 0.5
        filter_size_map: 0.5
        cube_side_length: 1000.0
        runtime_pos_log_enable: true
        map_file_path: "./test.pcd"

        common:
            lid_topic:  "/ouster/points"
            imu_topic:  "/ouster/imu"
            time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
            time_offset_lidar_to_imu: 0.0 # Time offset between lidar and IMU calibrated by other algorithms, e.g. LI-Init (can be found in README).
                                        # This param will take effect no matter what time_sync_en is. So if the time offset is not known exactly, please set as 0.0
            
        preprocess:
            lidar_type: 3                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
            scan_line: 128
            timestamp_unit: 3                 # 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
            blind: 3.0

        mapping:
            acc_cov: 0.1
            gyr_cov: 0.1
            b_acc_cov: 0.0001
            b_gyr_cov: 0.0001
            fov_degree:    180.0
            det_range:     170.0
            extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic
            extrinsic_T: [ 0.006253, -0.011775, 0.028535 ]
            extrinsic_R: [1., 0., 0.,
                        0., 1., 0.,
                        0., 0., -1.]

        publish:
            path_en:  false
            scan_publish_en:  true       # false: close all the point cloud output
            dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
            scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

        pcd_save:
            pcd_save_en: true
            interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                        # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

Ericsii, 2024-11-13:
看起来是fastlio没有正确订阅到lidar和imu的topic，请检查一下topic名字是否正确

tenmoonsninesky, 2024-11-22:
> This ROS package provide support for all Ouster sensors with FW v2.0 or later. Upon launch the driver will configure and connect to the selected sensor device, once connected the driver will handle incoming IMU and lidar packets, decode lidar frames and publish corresponding ROS messages on the topics of /ouster/imu and /ouster/points. In the case the used sensor supports dual return and it was configured to use this capability, then another topic will published named /ouster/points2 which corresponds to the second point cloud.

> 看起来是fastlio没有正确订阅到lidar和imu的topic，请检查一下topic名字是否正确

根据Ouster的ROS2驱动的介绍，其topic名字改为了`/ouster/imu`和`/ouster/points`，我在配置文件中也进行了如此修改（1），结果并没有变化。我还尝试故意修改了topic名字为错误的名字（2），事实证明尝试（1）和（2）都会导致在ROS接收到点云的一瞬间，Rviz中的Fixed Frame一项从warn变为error。

这导致该问题可能并不是因为topic名字错误产生的。于是我将ROS2的版本更换为了Humble，Ubuntu升级为22.04，这与推荐配置是一致的，但仍然会出现同样的错误。但我注意到，运行Fast-Lio时的INFO信息：

```
$ ros2 launch fast_lio mapping.launch.py config_file:=ouster128.yaml
[INFO] [launch]: All log files can be found below /home/guangdian/.ros/log/2024-11-22-16-50-56-598118-guangdian-Default-string-375000
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fastlio_mapping-1]: process started with pid [375018]
[INFO] [rviz2-2]: process started with pid [375020]
[fastlio_mapping-1] /home/guangdian/FASTLIO2/install/fast_lio/lib/fast_lio/fastlio_mapping: /home/guangdian/anaconda3/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /opt/ros/humble/lib/librclcpp.so)
[fastlio_mapping-1] /home/guangdian/FASTLIO2/install/fast_lio/lib/fast_lio/fastlio_mapping: /home/guangdian/anaconda3/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libvtkCommonCore-9.1.so.1)
[ERROR] [fastlio_mapping-1]: process has died [pid 375018, exit code 1, cmd '/home/guangdian/FASTLIO2/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/guangdian/FASTLIO2/install/fast_lio/share/fast_lio/config/ouster128.yaml --params-file /tmp/launch_params_yjbr1fmh'].
[rviz2-2] [INFO] [1732265457.004194476] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1732265457.004248873] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1732265457.029708047] [rviz2]: Stereo is NOT SUPPORTED
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-2] [INFO] [1732265494.904805192] [rclcpp]: signal_handler(signum=2)
[INFO] [rviz2-2]: process has finished cleanly [pid 375020]
```

中有这样一段提示：
`[ERROR] [fastlio_mapping-1]: process has died [pid 375018, exit code 1, cmd '/home/guangdian/FASTLIO2/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/guangdian/FASTLIO2/install/fast_lio/share/fast_lio/config/ouster128.yaml --params-file /tmp/launch_params_yjbr1fmh'].`

这句错误提示是在我开启ouster驱动之前就产生了。我不太清楚这里为什么会传两个相同的参数`--params-file` ，第一个是我改过的yaml，第二个似乎是一个缓存文件，且名字每一次都不同。是否是因为这里导致的问题？

tenmoonsninesky, 2024-11-22:
> 根据Ouster的ROS2驱动的介绍，其topic名字改为了`/ouster/imu`和`/ouster/points`，我在配置文件中也进行了如此修改（1），结果并没有变化。我还尝试故意修改了topic名字为错误的名字（2），事实证明尝试（1）和（2）都会导致在ROS接收到点云的一瞬间，Rviz中的Fixed Frame一项从warn变为error。 [...]

tenmoonsninesky, 2024-11-22:
> This ROS package provide support for all Ouster sensors with FW v2.0 or later. Upon launch the driver will configure and connect to the selected sensor device, once connected the driver will handle [...]

我刚刚修复了这个问题，可能是因为anaconda与C++环境冲突引起的，现在我可以运行FAST-LIO。

[image]

但是它似乎在轻微的移动之后发生严重漂移。

[image]

Ericsii, 2024-11-25:
漂移可能是imu外参和标定参数的问题

tenmoonsninesky, 2024-11-26:
> 漂移可能是imu外参和标定参数的问题

感谢，现在已经不会漂移了。
ouster的lidar-imu的旋转矩阵参数应该是
[-1,0,0
0,-1,0
0,0,1]

Ericsii, 2024-11-26:
done

Ericsii, 2024-11-26:
done

closed 2024-11-26 by Ericsii

hzlbbfrog, 2024-12-05:
您好！我遇到了同样的问题！开启Ouster-ros 驱动之后，Global status红了
[image]
并且同样提示
[image]

hzlbbfrog, 2024-12-05:
具体提示是[fastlio_mapping-1] [WARN] [1733424169.195025724] [laser_mapping]: New publisher discovered on topic '/ouster/imu', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
和您的问题一样。

hzlbbfrog, 2024-12-05:
我按照您的方案，输入是
ros2 launch ouster_ros sensor.launch.xml viz:=false point_cloud_frame:=os_sensor lidar_mode:=1024x10 point_type:=original use_system_default_qos:=false
但是这个问题依然存在，然后我查了一下sensor.launch.xml，use_system_default_qos这个默认值好像就false.

hzlbbfrog, 2024-12-05:
请问您有什么好的建议吗，如何让Global status显示正常，并且有点云出来呢。谢谢！

Ericsii, 2024-12-06:
> 我按照您的方案，输入是 ros2 launch ouster_ros sensor.launch.xml viz:=false point_cloud_frame:=os_sensor lidar_mode:=1024x10 point_type:=original use_system_default_qos:=false 但是这个问题依然存在，然后我查了一下sensor.launch.xml，use_system_default_qos这个默认值好像就false.

这里的问题是imu使用了非默认的QoS导致的，两种办法解决：
- 修改ouster的配置，使ouster的imu topic使用系统默认QoS
- 修改FAST_LIO代码中的这一行https://github.com/Ericsii/FAST_LIO_ROS2/blob/b6ebca01663793ad5faa55324d7af7576ed8bb2e/src/laserMapping.cpp#L929 
  为`sub_imu_ = this->create_subscription<sensor_msgs::msg::Imu>(imu_topic, 10, imu_cbk);`

hzlbbfrog, 2024-12-06:
您好！！非常感谢您的解答！
请问第二个解决方案，是修改哪里呢。
我使用第一个解决方案，
ros2 launch ouster_ros sensor.launch.xml viz:=false point_cloud_frame:=os_sensor lidar_mode:=1024x10 point_type:=original use_system_default_qos:=true之后，
有结果了，但是结果很奇怪，无法正常建图
[image]
然后lidar几乎不动，就出现这个
[image]
[image]

Ericsii, 2024-12-09:
@hzlbbfrog 应该和上面的问题一样，请修改lidar-imu外参
[-1,0,0
0,-1,0
0,0,1]

hzlbbfrog, 2024-12-11:
谢谢您！
然后昨天我试验的时候，突然报错
[image]
然后我重装了一下ubuntu，重新安装了之前的文件，还是报这个错，请问这个该咋办呢，搜了一下也没有好的解决方案 /(ㄒoㄒ)/~~

hzlbbfrog, 2024-12-11:
[fastlio_mapping-1] /home/hzl/fast_lio2_ws/install/fast_lio/lib/fast_lio/fastlio_mapping: symbol lookup error: /lib/aarch64-linux-gnu/libpcl_io.so.1.12: undefined symbol: libusb_set_option
[ERROR] [fastlio_mapping-1]: process has died [pid 17257, exit code 127, cmd '/home/hzl/fast_lio2_ws/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/hzl/fast_lio2_ws/install/fast_lio/share/fast_lio/config/ouster64_github.yaml --params-file /tmp/launch_params_v2k9347u'].

Ericsii, 2024-12-11:
看起来是你的libusb链接库被什么东西篡改了，请检查是不是使用了类似Anaconda之类的环境管理工具

hzlbbfrog, 2024-12-11:
谢谢博主！我搜了个方法，改一下环境变量就好了。
唉，不过飘移问题一直没有解决，试了好几种矩阵配置，都不行 /(ㄒoㄒ)/
一开始可以建出来，然后动一下雷达就飘了
[image]
[image]

Ericsii, 2024-12-11:
请用专门的外参工具来标定lidar-imu的外参和imu内参

hzlbbfrog, 2024-12-11:
好的好的我试试，谢谢您！！