Ericsii/FAST_LIO_ROS2 #25: Cannot to execute colcon build --symlink-install

maxxbolotov, 2024-11-26:
Hi!
There is an issue while i try to execute colcon build --symlink-install after all the previous steps.
Here is a log:
```
[5.304s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_package' doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[5.307s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_package' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> gtest
[5.316s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_lint' doesn't explicitly install a marker in the package index (colcon-ros currently does it implicitly but that fallback will be removed in the future)
[5.316s] WARNING:colcon.colcon_ros.task.ament_python.build:Package 'ament_lint' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)
--- stderr: gtest_vendor
CMake Error: The source directory "/opt/ros/humble/share/gtest_vendor" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/cmake/build.py", line 34, in build
    rc = await extension.build(
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 182, in _reconfigure
    buildfile = get_buildfile(cmake_cache)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 114, in get_buildfile
    if 'Visual Studio' in generator:
TypeError: argument of type 'NoneType' is not iterable
---
Failed   <<< gtest_vendor [0.04s, exited with code 1]
Aborted  <<< ament_lint [0.10s]                                                              
Aborted  <<< ament_package [0.10s]
Aborted  <<< gtest [0.15s]                                  

Summary: 0 packages finished [5.31s]
  1 package failed: gtest_vendor
  3 packages aborted: ament_lint ament_package gtest
  1 package had stderr output: gtest_vendor
  277 packages not processed
argument of type 'NoneType' is not iterable

```
As i understood it tries to find CMakeLists.txt at /opt/ros/humble/share/gtest_vendor but it doesn't exist there. I was thinking that such file would automatically appear after installing ROS2 Humble but its not.
So may be there is a way to install all required dirs in share manually or update already existing?

Ericsii, 2024-11-27:
What's your current workspace folder now? I didn't see anything related to FAST_LIO in your log.

If you installed ROS correctly, these packages (ament_package, ament_lint, gtest_vendor, gtest) should not appear in your building log.

maxxbolotov, 2024-11-27:
> What's your current workspace folder now? I didn't see anything related to FAST_LIO in your log.
> 
> If you installed ROS correctly, these packages (ament_package, ament_lint, gtest_vendor, gtest) should not appear in your building log.

My workspace folder is /opt/ros/humble/src
Here I clone FAST_LIO (/opt/ros/humble/src/FAST_LIO) 
Also here I installed ws_livox driver and it works properly. 

After I install ROS2 I launch listener-talker sample and it worked well. Rviz with ws_livox is also working. I can see everything that my Mid 360 sees.

Ericsii, 2024-11-27:
You should not put FAST_LIO in /opt/ros/humble/src. Please create a new folder somewhere else and put the source code into it

maxxbolotov, 2024-11-27:
> You should not put FAST_LIO in /opt/ros/humble/src. Please create a new folder somewhere else and put the source code into it

I reinstalled all the frameworks into the other folder as you said. Now its home/sageezz/src/ros_ws
This folder now contain ws_livox driver, Livox-SDK2 and here i cloned FAST_LIO.
rosdep install --from-paths ros_ws --ignore-src -y finished as it should.
After all the steps i still cannot execute colcon build --symlink-install
There is a log below:
```
~/src$ colcon build --symlink-install
[0.902s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/sageezz/src/ws_livox/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.902s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/opt/ros/humble/src/ws_livox/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.902s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/sageezz/ws_livox/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.902s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/sageezz/src_/ws_livox/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
[0.903s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sageezz/src/ws_livox/install/livox_ros_driver2' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.903s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/sageezz/src/ws_livox/install/livox_ros_driver2' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> livox_ros_driver2
Starting >>> Eigen3
Starting >>> livox_sdk2
Finished <<< livox_sdk2 [0.61s]                                                             
--- stderr: livox_ros_driver2                                                           
/usr/include/apr-1.0
apr-1
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/sageezz/src/ros_ws/ws_livox/src/livox_ros_driver2/LIVOX_INTERFACES_INCLUDE_DIRECTORIES
   used as include directory in directory /home/sageezz/src/ros_ws/ws_livox/src/livox_ros_driver2

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< livox_ros_driver2 [6.40s, exited with code 1]
Aborted  <<< Eigen3 [16.1s]                             

Summary: 1 package finished [16.9s]
  1 package failed: livox_ros_driver2
  1 package aborted: Eigen3
  2 packages had stderr output: Eigen3 livox_ros_driver2
  1 package not processed

```

Ericsii, 2024-11-27:
This is because your livox_ros_driver2 wasn't built correctly.

One way is to put livox_ros_driver2 in another workspace and compile it separately following the official repo's instructions.

Then run `source <livox_driver_ws>/install/setup.bash` command before building FAST_LIO2

closed 2024-12-09 by Ericsii