Ericsii/FAST_LIO_ROS2 #28: Ros 2 Jazzy

chaosrdd, 2024-12-10:
[image]
Hello. I'm new to using lidar sensors and I'd like to map with a livox 360, I'm on ubuntu 24.04 and I use ros jazzy, I followed the tutorial until the part asking me to colcon. At this point, a large number of errors appear, as in the example above. My livox sdk is well sourced, as is my ros. I wanted to know if anyone had a solution, I think it's a problem with ros Jazzy but I'm not sure.

Thank you in advance

Ericsii, 2024-12-11:
It looks like you're using a wrong C++ standard in the building. You can try to replace these lines in CMakeLists.txt with the latter block, which may help.

https://github.com/Ericsii/FAST_LIO_ROS2/blob/b6ebca01663793ad5faa55324d7af7576ed8bb2e/CMakeLists.txt#L8-L15

```c++
 ADD_COMPILE_OPTIONS(-std=c++17) 
 ADD_COMPILE_OPTIONS(-std=c++17) 
 set(CMAKE_CXX_FLAGS "-std=c++17 -O3") 
  
 add_definitions(-DROOT_DIR=\"${CMAKE_CURRENT_SOURCE_DIR}/\") 
  
 set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fexceptions") 
 set(CMAKE_CXX_STANDARD 17) 
```

chaosrdd, 2024-12-12:
Thanks you it work

closed 2024-12-12 by chaosrdd