Ericsii/FAST_LIO_ROS2 #31: Better way to perform navigation tasks and LiDAR orientation with Mid360

effibot, 2025-01-29:
Hi, first of all, I would thank you for your work. I was able to make a map using a mid360, but I would like to ask two questions.

1. For my use-case, I have to use two lidars (two mid360), orienting the sensor to face the ground. Basically, I need to invert the Z-axis. I have to make some changes to the code, or to some configuration, to perform a correct mapping?
2.  As for navigation, could you suggest a way to perform path-planning and obstacle avoidance missions? In my previous projects I always used NAV2, but it requires a 2D map in order to produce a costmap. I thought to use some pointcloud-to-laserscan methods and then pass it through a classic 2D-Slam algorithm, but I would like to use the full-potential of this 3D sensor.

Thanks for your help.

Ericsii, 2025-02-24:
The official NAV2 now support pointcloud as input to generate costmap. You can refer to its document. Use the VoxelMap plugin.

https://docs.nav2.org/configuration/packages/costmap-plugins/voxel.html

effibot, 2025-02-26:
Thanks for your reply. In the meanwhile I succeded in the setup of a 2D map with the Octomap Package, using camera_init as fixed frame to generate a projected_map, but i couldn't generate costmaps with NAV2 using the neither the voxel-layer nor the STVL plugin. 
I got a warning from the global_costmap that tells that the robot is outside the bounds of the map. 
If you ever succeded in the generation of a costmap with FAST_LIO and have other suggestions I'm glad to hear you. Meanwhile i'll close the issue and i will open another to NAV2.

closed 2025-02-26 by effibot