Ericsii/FAST_LIO_ROS2 #32: No PCD saved

Joaquin-Navas, 2025-02-07:
Hi,

I'm succesfully running `ros2 launch fast_lio mapping.launch.py config_file:=mid360.yaml` and getting the map in rviz.

Issue is that no point cloud is saved in PCD.

3.4 PCD file save indicates to change pcd_save_enable in launchfile to 1, but I'm not able to find nor to set the launch file.

Thanks.

xiaobaigao36, 2025-02-24:
me too,have you solved it now?

Ericsii, 2025-02-24:
The pcd saving process now needs to be triggered from outside. There is a `map_save` service you can use the service caller in rqt to trigger this.

closed 2025-02-24 by Ericsii

20-wash, 2025-02-24:
Nono you can so this, there is a code block which is commented ,you need to uncomment that one 
_____________________________________

To activate the above edit uncomment the function below ::::::::::

/************* save map **************/
/* 1. make sure you have enough memories
/* 2. noted that pcd save will influence the real-time performences **/

[image]

Joaquin-Navas, 2025-02-25:
> Nono you can so this, there is a code block which is commented ,you need to uncomment that one On doing that, every time you press ctrl+c it saves with the same name. And also overwrites
> 
> To activate the above edit uncomment the function below ::::::::::
> 
> /************* save map ************__/ /_ 1. make sure you have enough memories /_ 2. noted that pcd save will influence the real-time performences **/
> 
> [image]

This is what I did and works.

I guess @Ericsii fix is more elegant though.

ilisparrow, 2025-10-24:
Hello, 
Thank you both for the reply. 
A small additiont hat might help others : 
A new parameters needs to be added to in your yaml file
Also, th efile is saved in ./test.pcd

        publish:
            path_en:  false
            scan_publish_en:  true       # false: close all the point cloud output
            dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
            scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame
            map_en: true                 # ADD THIS LINE

Thank you @Ericsii for the porting !