Ericsii/FAST_LIO_ROS2 #34: build issue

marapakabannudeepak, 2025-02-24:
Eigen:/usr/include/eigen3
CMake Error at CMakeLists.txt:95 (add_executable):
  Cannot find source file:

    include/ikd-Tree/ikd_Tree.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:95 (add_executable):
  No SOURCES given to target: fastlio_mapping

Ericsii, 2025-02-24:
Because you did not clone the submodule together

marapakabannudeepak, 2025-02-24:
> Because you did not clone the submodule together

How to fix  that issue?

Ericsii, 2025-02-24:
There are two ways to clone the submodule

- When the first time cloning the repo, add argument --recursive. 
  `git clone https://github.com/Ericsii/FAST_LIO_ROS2.git --recursive`
- If you already cloned the repo, run the following two commands in the source folder.
  `git submodule init`
  `git submodule update`

closed 2025-02-24 by Ericsii