Ericsii/FAST_LIO_ROS2 #36: mid360无法作mapping和保存地图

jonathantyl97, 2025-02-25:
@Ericsii 
首先非常感谢你做这个Repo， 我已成功用ROS2去打开。

可是我现在遇到了以下问题：

- mid360可以进行扫描，也看得到点云，可是点云地图不会保存，pcd地图也不会保存。
- 我已经uncomment在laserMapping.cpp里的保存地图function。
- 打开了launch文件后，扫描一直在一个定点旋转，并不会有trajectory，尽管IMU已打开。

[image]

mid360.yaml:
`
        common:
            lid_topic: "/livox/lidar"
            imu_topic: "/livox/imu"
            time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
            time_offset_lidar_to_imu: 0.0 # Time offset between lidar and IMU calibrated by other algorithms, e.g. LI-Init (can be found in README).
                                        # This param will take effect no matter what time_sync_en is. So if the time offset is not known exactly, please set as 0.0

        preprocess:
            lidar_type: 1              
            scan_line: 4                
            blind: 0.5
            timestamp_unit: 3           # 3 for microseconds
            scan_rate: 10

        mapping:
            acc_cov: 0.1              # Reduced from 0.1
            gyr_cov: 0.1              # Reduced from 0.1
            b_acc_cov: 0.0001
            b_gyr_cov: 0.0001
            det_range: 50.0          # Reduced for better local mapping
            fov_degree: 180.0
            extrinsic_est_en: true
            extrinsic_T: [ 0.0, 0.0, 0.0 ]    # Default to zero if you don't have calibration
            extrinsic_R: [ 1., 0., 0.,
                          0., 1., 0.,
                          0., 0., 1.]

        publish:
            path_en: true                # true: publish Path
            effect_map_en: false         # changed to true: publish Effects
            map_en: true                 # true: publish Map cloud
            scan_publish_en: true        # true: publish point cloud output
            dense_publish_en: false       # true: publish dense point clouds
            scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame`

jonathantyl97, 2025-02-25:
@Ericsii https://drive.google.com/file/d/1Cwp29qvHg9128-An2TyJkhR_1vaoUy4h/view?usp=sharing

是这种情况

jonathantyl97, 2025-02-25:
我没办法保存地图，因为似乎没有点云被保存

Ericsii, 2025-02-25:
这里看起来像lio并没有正常运行，能看一下输出的log吗

jonathantyl97, 2025-02-25:
-19-868029-jonathan-Dell-G16-7630-72013
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [72014]
[INFO] [static_transform_publisher-2]: process started with pid [72016]
[INFO] [static_transform_publisher-3]: process started with pid [72018]
[INFO] [static_transform_publisher-4]: process started with pid [72020]
[INFO] [livox_ros_driver2_node-5]: process started with pid [72022]
[INFO] [fastlio_mapping-6]: process started with pid [72024]
[INFO] [rviz2-7]: process started with pid [72026]
[static_transform_publisher-1] [WARN] [1740468319.903617390] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [WARN] [1740468319.903727849] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [WARN] [1740468319.903856963] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-4] [WARN] [1740468319.904274068] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-1] [INFO] [1740468319.912296447] [static_tf_map_odom]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'map' to 'camera_init'
[static_transform_publisher-2] [INFO] [1740468319.913842383] [static_tf_odom_base]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'camera_init' to 'base_link'
[static_transform_publisher-4] [INFO] [1740468319.914470863] [static_tf_lidar_imu]: Spinning until stopped - publishing transform
[static_transform_publisher-4] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-4] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-4] from 'livox_frame' to 'imu_link'
[livox_ros_driver2_node-5] [INFO] [1740468319.915036174] [livox_lidar_publisher]: Livox Ros Driver2 Version: 1.2.4
[static_transform_publisher-3] [INFO] [1740468319.915179076] [static_tf_base_lidar]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'base_link' to 'livox_frame'
[livox_ros_driver2_node-5] [INFO] [1740468319.915320409] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros_driver2_node-5] [INFO] [1740468319.915330384] [livox_lidar_publisher]: Config file : /home/jonathan/Toyota/ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/config/MID360_config.json
[livox_ros_driver2_node-5] LdsLidar *GetInstance
[livox_ros_driver2_node-5] config lidar type: 8
[livox_ros_driver2_node-5] successfully parse base config, counts: 1
[livox_ros_driver2_node-5] [INFO] [1740468319.916743521] [livox_lidar_publisher]: Init lds lidar success!
[fastlio_mapping-6] [INFO] [1740468319.948447542] [fast_lio]: p_pre->lidar_type 3
[fastlio_mapping-6] Multi thread started 
[fastlio_mapping-6] ~~~~/home/jonathan/Toyota/ws_livox/src/FAST_LIO_ROS2/ file opened
[fastlio_mapping-6] [INFO] [1740468319.950190385] [fast_lio]: Node init finished.
[rviz2-7] [INFO] [1740468320.099066248] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-7] [INFO] [1740468320.099159624] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-7] [INFO] [1740468320.109720913] [rviz2]: Stereo is NOT SUPPORTED
[livox_ros_driver2_node-5] GetFreeIndex key:livox_lidar_3288443072.
[livox_ros_driver2_node-5] Init queue, real query size:16.
[livox_ros_driver2_node-5] Lidar[0] storage queue size: 10
[livox_ros_driver2_node-5] set pcl data type, handle: 3288443072, data type: 1
[livox_ros_driver2_node-5] set scan pattern, handle: 3288443072, scan pattern: 0
[livox_ros_driver2_node-5] begin to change work mode to 'Normal', handle: 3288443072
[livox_ros_driver2_node-5] successfully set data type, handle: 3288443072, set_bit: 2
[livox_ros_driver2_node-5] successfully set pattern mode, handle: 3288443072, set_bit: 0
[livox_ros_driver2_node-5] successfully set lidar attitude, ip: 192.168.1.196
[livox_ros_driver2_node-5] successfully change work mode, handle: 3288443072
[livox_ros_driver2_node-5] successfully enable Livox Lidar imu, ip: 192.168.1.196
[livox_ros_driver2_node-5] [INFO] [1740468322.917053442] [livox_lidar_publisher]: livox/imu_192_168_1_196 publish use imu format
[livox_ros_driver2_node-5] [INFO] [1740468322.918491657] [livox_lidar_publisher]: livox/lidar_192_168_1_196 publish use PointCloud2 format

jonathantyl97, 2025-02-25:
这里是我的launch文件：

`   livox_driver_node = Node(
        package='livox_ros_driver2',
        executable='livox_ros_driver2_node',
        name='livox_lidar_publisher',
        parameters=[{
            'xfer_format': 0,
            'multi_topic': 1,
            'data_src': 0,
            'publish_freq': 10.0,
            'output_data_type': 0,
            'frame_id': 'livox_frame',
            'user_config_path': user_config_path,
            'enable_fan': True,
            'enable_imu': True,
            'return_mode': 0,
            'imu_rate': 200,
            'host_net_info': {
                'host_ip': '192.168.1.50',
                'host_port': 56100,
                'subnet_mask': '255.255.255.0',
                'dev_ip': '192.168.1.196'
            }
        }],
        output='screen'
    )

    fast_lio_node = Node(

        package='fast_lio',
        executable='fastlio_mapping',
        name='fast_lio',
        parameters=[
            config_path,
            {
                'common.time_sync_en': True,
                'common.time_offset_lidar_to_imu': 0.0,
                'preprocess.lidar_type': 3,
                'preprocess.scan_line': 4,
                'point_cloud_topic': '/livox/lidar',
                'imu_topic': '/livox/imu',
                'publish_point_cloud': True,
                'publish_odometry': True,
                'save_trajectory': True,
                'path_pub_period': 0.02,
                'frame_id': 'camera_init',
                'publish_global_map': True,
                'global_map_pub_period': 1.0,
                'max_iteration': 3,
                'filter_size_surf': 0.3,
                'filter_size_map': 0.2,
                'cube_side_length': 1000.0,
                'feature_extract_enable': True,
                'mapping.fov_degree': 180.0,
                'mapping.det_range': 50.0,
                'mapping.map_resolution': 0.2,
                'mapping.max_points_per_cell': 5,
                'mapping.min_distance': 0.15,
                'publish_rate_hz': 10.0,
                'mapping.acc_cov': 0.1,
                'mapping.gyr_cov': 0.1,
                'pcd_save.pcd_save_en': True,
                'pcd_save.interval': 1,
                'pcd_save.save_dir': './map/',
            }
        ],
        output='screen'
    )
`

Ericsii, 2025-02-25:
有试过直接使用repo里的launch文件来运行吗？我不清楚为什么这里会有static_transform_publisher出现

jonathantyl97, 2025-02-28:
我确实在launch那边有些失误。可以close这个issue了。谢谢你的帮助!

closed 2025-02-28 by jonathantyl97