Ericsii/FAST_LIO_ROS2 #39: livox_ros_driver not found

YanhaoZhang, 2025-03-12:
Thanks for releasing the code. I am following the instruction, and trying to play a rosbag to see if the compile was successfully. However, I find the the rviz is black without anything. To be more specific, what I did was

1. install ROS Humble
2. Install driver based on https://github.com/Livox-SDK/livox_ros_driver2
3. Build FAST_LIO_ROS2 
4. Launch the code by 

```shell
ros2 launch fast_lio mapping.launch.py config_file:=avia.yaml
ros2 bag play my_bag_path
```

I also check with the issue [here](https://github.com/Ericsii/FAST_LIO_ROS2/issues/19), however, my problem was still there. 

I guess it might be because of the converted bag file also uses ros1 format, as the following warning:

[WARN] [1741776226.045059245] [rosbag2_player]: Ignoring a topic '/rosout', reason: Something went wrong loading the typesupport library for message type rosgraph_msgs/Log. Library could not be found..
[WARN] [1741776226.046921336] [rosbag2_player]: Ignoring a topic '/livox/lidar', reason: package 'livox_ros_driver' not found, searching: [/home/baai/ws_livox/install/livox_ros_driver2, /opt/ros/humble].

I was wondering if you could please give me some suggestions?
The rosbag I used was `2020-09-16-quick-shack`, and I used the following way to convert it:
```shell
rosbags-convert --src ~/xxx.bag --dst ~/output_folder
```

Btw, I find that the link you share the rosbag (https://drive.google.com/drive/folders/1VBK5idI1oyW0GC_I_Hxh63aqam3nocNK?usp=sharing) and that for the python script are not accessible. I was wondering if you could please share it again? 

Thanks.

Ericsii, 2025-03-17:
This is because I changed the dependency of livox_ros_driver to livox_ros_driver2. But the converted rosbag is still using the old version livox_ros_driver, which doesn't match the requirement.

In order to adapt the old version you need to modify the source code in CMakeLists and message type in laserMapping.cpp.

Ericsii, 2025-04-16:
This issue may give some help. https://github.com/hku-mars/FAST_LIO/issues/373