Ericsii/FAST_LIO_ROS2 #47: Can't See Camera Frame in RViz2 and No Registered Cloud Published

3bdul1ah, 2025-05-07:
Hey, I'm using FastLIO with a Livox Mid-360 on ROS 2 Foxy (Ubuntu 20.04).

At first, I ran into the common "placeholder error" when building the workspace. I fixed it by editing the `LaserMapping` file, as mentioned in the GitHub issues. After that, everything built fine.

Now, the problem is when I launch the mapping node and the Livox Mid-360 launch file, I get a black screen in RViz2—no camera init frame or point cloud appears.

I tried echoing `/livox/lidar` and `/livox/imu`, and both are publishing fine. The frame being used is `livox_frame`. But when I echo the registered point cloud topic, nothing shows up.

Any idea what might be wrong or what I should check?

Thanks!

Ericsii, 2025-05-08:
Can you provide logs when you launch FASTLIo?

3bdul1ah, 2025-05-08:
## FAST_LIO on ROS 2 Foxy

Thanks for your reply

Actually, I just managed to solve the issue, and it's working now. Here's what I did:

When trying to publish custom messages, I was getting some errors with the original livox_ros_driver. So I used your fork instead:

```bash
cd <ros2_ws>/src
git clone -b feature/use-standard-unit https://github.com/Ericsii/livox_ros_driver2.git
cd ..
colcon build --symlink-install
source install/setup.bash
```

Then launched it:

```bash
ros2 launch livox_ros_driver2 msg_MID360_launch.py
```

Checked the custom topic, and it was publishing just fine this time.

Next, I cloned your FAST_LIO fork:

```bash
cd <ros2_ws>/src
git clone --recursive https://github.com/Ericsii/FAST_LIO_ROS2.git
cd ..
colcon build --symlink-install
```

Ran into a small error in `laserMapping.cpp`, but fixed it using the solution from this issue:

https://github.com/hku-mars/FAST_LIO/issues/234#issuecomment-2244430899

After that:

```bash
colcon build --symlink-install
source install/setup.bash
```

Launched FAST_LIO:

```bash
ros2 launch fast_lio mapping.launch.py config_file:=mid360.yaml
```

And everything worked perfectly

[image]

Thanks again, for your forks 🙏🏼

closed 2025-05-08 by Ericsii