Ericsii/FAST_LIO_ROS2 #52: 使用.sh开机自启动的脚本就会出现跟你一样的问题，，我查了fastlio的日志给出的问题是 Leaf size is too small问题

You202373, 2025-07-13:
> [INFO] [launch]: All log files can be found below /home/sunrise/.ros/log/2025-07-06-21-14-25-482771-ubuntu-17312
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [fastlio_mapping-1]: process started with pid [17405]
> [fastlio_mapping-1] [INFO] [1751807666.574100385] [laser_mapping]: p_pre->lidar_type 1
> [fastlio_mapping-1] Multi thread started 
> [fastlio_mapping-1] ~~~~/home/sunrise/ros_ws/src/FAST_LIO_ROS2/ file opened
> [fastlio_mapping-1] [INFO] [1751807666.625215097] [laser_mapping]: Node init finished.
> [fastlio_mapping-1] [WARN] [1751807668.811555168] [laser_mapping]: No point, skip this scan!
> [fastlio_mapping-1] 
> [fastlio_mapping-1] [WARN] [1751807668.823507396] [laser_mapping]: No point, skip this scan!
> [fastlio_mapping-1] 
> [fastlio_mapping-1] [WARN] [1751807668.831577243] [laser_mapping]: No point, skip this scan!
> [fastlio_mapping-1] 
> [fastlio_mapping-1] [WARN] [1751807668.841573635] [laser_mapping]: No point, skip this scan!
> [fastlio_mapping-1] 
> [fastlio_mapping-1] IMU Initial Done
> [fastlio_mapping-1] [WARN] [1751807668.852113486] [laser_mapping]: No point, skip this scan!
> [fastlio_mapping-1] 
> [fastlio_mapping-1] [INFO] [1751807668.963967130] [laser_mapping]: Initialize the map kdtree
> [ERROR] [fastlio_mapping-1]: process has died [pid 17405, exit code -11, cmd '/home/sunrise/ros_ws/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/sunrise/ros_ws/install/fast_lio/share/fast_lio/config/mid360.yaml --params-file /tmp/launch_params_0r1dh95c'].
> 这是我今天刚启动就退出的
> 我的launch livox_ros_driver2 msg_MID360_launch.py publish_freq:=30.0在20s前启动的。
> 而且这是概率性问题，有时它可以持续运行数十分钟 

 _Originally posted by @dddyyyy123 in [#51](https://github.com/Ericsii/FAST_LIO_ROS2/issues/51#issuecomment-3041500067)_

dddyyyy123, 2025-07-13:
请问你解决了吗

  

 [fastlio_mapping-1] [INFO] [1751807666.574100385] [laser_mapping]: p_pre-&gt;lidar_type 1

dddyyyy123, 2025-07-13:
我怀疑是我的点云频率只有设定值的1/8导致的

You202373, 2025-07-16:
@dddyyyy123 我这边解决了

dddyyyy123, 2025-10-20:
时间同步关掉，或者等时间同步完再启动

 
请问是怎么解决的，急求解决办法